EddyVerbruggen/nativescript-fingerprint-auth #2: Android Fingerprint Scanners Support

tsvetan-ganev, 2016-06-04:
Is support for Android M's fingerprint scanning features planned?

EddyVerbruggen, 2016-06-04:
Not really as I don't have a compatible device. Do you have one?

tsvetan-ganev, 2016-06-04:
Yes. I just bought a Nexus 5X. The problem is I don't have a USB-C to USB-A cable yet. But as soon as I get one, I will start experimenting with the Android Finger Scanner API. If I come up with a way to use it with NativeScript I will contact you.

EddyVerbruggen, 2016-06-04:
That's supercool, keep me posted :)

iDev0urer, 2016-07-11:
Big +1 on this. If you need any help let me know what I can do. My experience with developing nativescript plugins is 0, but I too have a Nexus 5X and would like this feature

tsvetan-ganev, 2016-07-16:
@EddyVerbruggen, @iDev0urer, I had some spare time and came up with [the following sample project](https://github.com/tsvetan-ganev/nativescript-android-confirm_credential). It is based on a Google sample and uses the Confirm Credential API which means that users can authenticate with PIN, Pattern,  Password and Fingerprint scanner (if they have set up one). Is this going to integrate well with your existing plugin or should I look at the standalone Android Fingerprint Scanner APIs instead?

iDev0urer, 2016-07-16:
Sounds to me like something that might be better standalone, but good job!

EddyVerbruggen, 2016-11-29:
Rebranded this plugin: `nativescript-touchid` ➡ `nativescript-fingerprint-auth` to accommodate for having added Android support. Version 3.0.0 is now live on npm.

closed 2016-11-29 by EddyVerbruggen