EddyVerbruggen/nativescript-fingerprint-auth #4: FingerPrint shows strange on android for HuaWei or Xiaomi

ITonly, 2016-12-19:
The fingerprint for Samsung（SM-G9280） is ok,the screenshot is as belows:
[image]
But  for HuaWei or Xiaomi ,it enters pin code page, while "Touch ID is available",which confuses me, the result is as follows on Huawei Note8 ( Huawei EMUI4.1+Android 6.0)

[image]
How can I fix the problem?

svzi, 2017-01-30:
I can confirm that it doesn't work on Xiaomi devices. At least not on my Xiaomi Redmi PRO, which is running Android 6.0.1 and is offering a working fingerprint sensor.

Using your plugin will only prompt for my device PIN, instead of my fingerprint.

acidbiscuit, 2019-01-12:
Same behavior Xiaomi Mi6.  

Setting `useCustomAndroidUI: true` in `verifyFingerprint` options resolved the issue with basic fingerprint authentication modal. No additional configurations made.  
This way execution passes [this](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/master/src/fingerprint-auth.android.ts#L95) line and launches `fingerPrintManager`.

If condition `useCustomAndroidUI` is not set, device PIN (not fingerprint) will be invoked on [this](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/master/src/fingerprint-auth.android.ts#L180) line (with `showAuthenticationScreen`) method.