EddyVerbruggen/nativescript-fingerprint-auth #5: Custom UI for Fingerprint Auth

chimon2000, 2017-01-09:
Would like to be able to use this plugin, but with a UI similar fashion to [FingerprintDialog](https://github.com/googlesamples/android-FingerprintDialog).

It'd be great to be able to specify a custom UI, and whether that UI that can show up in a dialog.

kalininskiy, 2017-09-06:
+1
I want disable PINCODE in Android fingerprint dialog screen, for only show fingerprint request and cancel button.

teerarak, 2017-10-10:
I want Use PINCODE IN MyApplication

EddyVerbruggen, 2018-01-26:
Hey folks,

Thanks for this request! I wasn't even aware Android allowed customizing the default screen this plugin currently uses.

I'm working on adding a custom UI that's more user friendly, based on the repo @chimon2000 linked to. It's shaping up nicely and will be added to this plugin soon:

[image]

EddyVerbruggen, 2018-01-30:
With plugin version 6.0.0 you can now pass in `useCustomAndroidUI: true` to trigger the UI the OP posted. See https://github.com/EddyVerbruggen/nativescript-fingerprint-auth#verifyfingerprint for details.

closed 2018-01-30 by EddyVerbruggen