EddyVerbruggen/nativescript-fingerprint-auth #7: "touchid.available()" inside emulator returns false

Alxblsk, 2017-02-05:
I'm trying to run the example of code for Android 7.0 emulator. I set up a fingerprint and added the following code the the project

```var touchid = require("nativescript-touchid");
var observableModule = require("data/observable");
 
function pageLoaded(args) {
    var page = args.object;
    var data = new observableModule.Observable({authenticated: false});
    touchid.available().then(function(avail) {
        if(avail) {
            touchid.verifyFingerprint({
                message: "Scan with your finger",
                fallbackTitle: "Enter PIN"
            }).then(function() {
                data.authenticated = true;
            }, function(error) {
                console.log("Fingerprint NOT OK" + (error.code ? ". Code: " + error.code : ""));
            });
        }
    });
    page.bindingContext = data;
}
 
exports.pageLoaded = pageLoaded;```

However, console displays that Fingerprint is not available: 

```
JS: Avail:  false
JS: Fingerprint NOT OK
```

I suspect that something is wrong here. I'm using API v.24.
It would be great if you take a look on this issue. Thanks in advance!

EddyVerbruggen, 2017-02-05:
Which Android emulator that you're using supports fingerprint? AFAIK it needs the hardware to work..

Alxblsk, 2017-02-06:
This is standard emulator from Android studio. It has an opportunity to touch sensor. It looks like 
[image]

It works great with native login screen, but doesn't work inside NativeScript application. I'm just wondering, whether we can make it work.

erjdriver, 2017-04-22:
Works fine!

You first have to go to phone settings and add a "fingerprint".

StevenLJackson1, 2017-05-11:
@erjdriver, that was my experience, as well.

closed 2017-11-06 by EddyVerbruggen