EddyVerbruggen/nativescript-fingerprint-auth #10: iOS verifyFingerprintWithCustomFallback() method fallback error alert freezes the UI

Jaishankarb, 2017-07-05:
we have used this plugin to login using the touch id and we are facing the following issue 

when i call the function  verifyFingerprintWithCustomFallback() on load for the success scenario it works but in iOS after continuous failed attempt the Biometry is locked so when you call this function on load it executes the error i am just triggering the alert in this error block.

now the alert window freezes the UI please see the attached Screen shot  

` this.fingerprintAuth.verifyFingerprintWithCustomFallback({
      message: 'Sign in with the online ID ' + this.userName.substring(0, 3) + this.userName.substring(3).replace(/./g, '*'), // optional
      fallbackMessage: 'Password' // optional
    }).then(
      () => {
        this.navHome();
      },
      (error) => {
            dialogs.alert({
              title: "`Touch ID is locked out`",
              message: ("please lock the device and unlock the device"),
              okButtonText: "Ok"
            });
          
        }
      });
  }`

[image]

EddyVerbruggen, 2017-07-05:
Add a little timeout before showing the alert. There can be one 1 alert at a time on ios.

Jaishankarb, 2017-07-05:
thanks on set time out it is working as expected