EddyVerbruggen/nativescript-fingerprint-auth #12: Android: Cipher error

ownleviathan, 2017-08-18:
Hello
During my development, when I do the touchid, I need to do several times the touch to get access.

Finally, I got and error:

S: touch?true
JS: Error: javax.crypto.AEADBadTagException
JS:     android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:517)
JS:     javax.crypto.Cipher.doFinal(Cipher.java:2056)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1021)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:903)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:890)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:874)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:866)
JS:     com.tns.NativeScriptActivity.onStart(NativeScriptActivity.java:38)
JS:     android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1256)
JS:     android.app.Activity.performStart(Activity.java:6959)
JS:     android.app.Activity.performRestart(Activity.java:7075)
JS:     android.app.Activity.performResume(Activity.java:7080)
JS:     android.app.ActivityThread.performResumeActivity(ActivityThread.java:3768)
JS:     android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3832)
JS:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1681)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:154)
JS:     android.app.ActivityThread.main(ActivityThread.java:6682)
JS:     java.lang.reflect.Method.invoke(Native Method)
JS:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1534)
JS:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1424)
JS: Caused by: android.security.KeyStoreException: Signature/MAC verification failed
JS:     android.security.KeyStore.getKeyStoreException(KeyStore.java:1107)
JS:     android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.update(KeyStoreCryptoOperationChunkedStreamer.java:132)
JS:     android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.doFinal(KeyStoreCryptoOperationChunkedStreamer.java:217)
JS:     android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:506)
JS:     javax.crypto.Cipher.doFinal(Cipher.java:2056)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1021)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:903)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:890)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:874)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:866)
JS:     com.tns.NativeScriptActivity.onStart(NativeScriptActivity.java:38)
JS:     android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1256)
JS:     android.app.Activity.performStart(Activity.java:6959)
JS:     android.app.Activity.performRestart(Activity.java:7075)
JS:     android.app.Activity.performResume(Activity.java:7080)
JS:     android.app.ActivityThread.performResumeActivity(ActivityThread.java:3768)
JS:     android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3832)
JS:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1681)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:154)
JS:     android.app.ActivityThread.main(ActivityThread.java:6682)
JS:     java.lang.reflect.Method.invoke(Native Method)
JS:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1534)
JS:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1424)
JS: << UIApplication: com.tns.NativeScriptActivity@9a13e33 resume >>
JS: RDY TO RESUME>>>>>>>>>>>>>:
JS: IsSignedId in resume with SUSPENDED: true
JS: Enter in touchIdGoHome resume
JS: touch?true
JS: Error: javax.crypto.AEADBadTagException
JS:     android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:517)
JS:     javax.crypto.Cipher.doFinal(Cipher.java:2056)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1021)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:903)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:890)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:874)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:866)
JS:     com.tns.gen.android.app.Application_ActivityLifecycleCallbacks.onActivityResumed(Application_ActivityLifecycleCallbacks.java:24)
[... 77 lines cut ...]
    "nativescript-intl": "3.0.0",
    "nativescript-loading-indicator": "2.3.2",
    "nativescript-master-technology": "1.1.1",
    "nativescript-oauth": "file:../nativescript-oauth",
    "nativescript-sqlite": "^1.1.7",
    "nativescript-statusbar": "1.0.0",
    "nativescript-telerik-ui": "2.0.1",
    "rxjs": "5.4.1",
    "tns-core-modules": "3.0.1",
    "utf8": "2.1.2",
    "zone.js": "0.8.12"
  },
  "devDependencies": {
    "babel-traverse": "6.22.0",
    "babel-types": "6.25.0",
    "babylon": "6.17.4",
    "filewalker": "0.1.3",
    "lazy": "1.0.11",
    "nativescript-dev-typescript": "0.4.5",
    "tns-platform-declarations": "3.0.1",
    "tslint": "5.4.3",
    "typescript": "2.3.4"
  }
}
Thanks

lsowmya, 2017-09-27:
hello,
I am facing the same issue, where on resume, when I do the touchid it keeps on repeating several times and then I get access.

Is there any solution for this issue?

TheJimmyPan, 2017-12-08:
Hello,
I am having a similar issue. Has anyone found a solution?
I am using this.fingerprintAuth.verifyFingerprint inside of an application.AndroidApplication.activityResumedEvent and when it is checking for a fingerprint I get.

JS: >>> onActivityResult 1: 1
JS: >>> onActivityResult 2: -1
JS: >>> android.app.Activity.RESULT_OK: -1
JS: else Hvs_GCrS-sn43EY7OafmNkjHlMjyrGOH-s9A5QDL 
JS: >>> onActivityResult 1: 1
JS: >>> onActivityResult 2: -1
JS: >>> android.app.Activity.RESULT_OK: -1
JS: >>> onActivityResult 1: 1
JS: >>> onActivityResult 2: -1

Is there a different approach for checking fingerprint every time a user tries to access the application?

EddyVerbruggen, 2017-12-14:
@TheJimmyPan That log is perfectly normal in case your started fingerprint authentication 3 times while that log was printed.

@ all: Fingerprints on Android seem to be a bit of hit and miss on various phones. The only supported phone I have is a Samsung S8 which works just fine - which devices do you own the work/don't work?

EddyVerbruggen, 2017-12-14:
I make a small change of which I'm quite sure won't help, but let's give 5.1.0 a try anyway..

Note that if the problem persists I won't be able to fix it because my Samsung S8 behaves correctly. So you'll need to try and fix it yourself.

closed 2017-12-14 by EddyVerbruggen

TheJimmyPan, 2017-12-14:
@EddyVerbruggen Actually what's happening is I am trying to get the auth token after fingerprint and after the fingerprint triggers the first time it gives me the token and then it gives me the fingerprint screen again and then I am stuck in an infinite loop trying to get out of the fingerprint screen.