EddyVerbruggen/nativescript-fingerprint-auth #15: this.fingerprintAuth.verifyFingerprintWithCustomFallback , not able to redirect page on failure

VimalKumarS, 2017-10-05:
Hi ,

I am using Toch id plugin, works great when successfully scan the touch id, but on failure to scan i m trying to redirect the route, it fails to do so,  and app UI freeze,, even i tried set time out, it doesn't work.
Any suggestion?

thanks
vimal

Vishnu0306, 2017-11-20:
We are using "nativescript-fingerprint-auth" library  5.0.0 version, i have tried invalid fingerprint for first and second times showing "Try Again" popup message, For third time library was return -1 error code, Even i continue invalid fingerprint for 4th and 5th time it was show "Try Again" popup. Again i tried for 6th time library was return -8, even i tried 7th time library return "undefined". Then after when i try invalid fingerprint its always return "undefined".

We need after tried third times reset fingerprint object. 

PFB code here i have initialized fingerprint object on two place.

`constructor( ) {
        this.fingerprintAuth = new FingerprintAuth();
}
this.fingerprintAuth.available().then((avail: any) => {
            if (avail) {
                this.fingerprintAuth.verifyFingerprintWithCustomFallback({
                    message: "Scan your finger",
                    fallbackMessage: "Decline"
                }).then(
                    () => {},
                    (error) => {
if (error.code === -1 ) {
this.fingerprintAuth = new FingerprintAuth();
}
}
}
}`
                        
Tested Device : iphone 6.

EddyVerbruggen, 2017-11-20:
iOS / Android?
Platform versions?
Sim / Device?
What's the code exactly? Especially inside the failure callback.

erjdriver, 2018-01-19:
I see then when the user taps on the fallback message (ENTER PASSCODE) - the Promise() fails with a code of -3.

How do I get iOS to display the enter password screen?

Thanks.

closed 2018-01-20 by EddyVerbruggen