EddyVerbruggen/nativescript-fingerprint-auth #19: verifyFingerprint without white screen

EduhCosta, 2017-12-13:
How can I just call the verifyFingerprint method without having to see an auxiliary screen ??
I would like it to just show up a dialog telling you the functionality, not the white screen.

Can someone help me??

Thank`s

EddyVerbruggen, 2017-12-13:
What's the white screen? Can you attach a little video perhaps?

EduhCosta, 2017-12-13:
This screen, 

[image]

I wish it did not just show an alert, waiting for the validation.

I`m called this method from see this screen:
`public doVerifyFingerprintWithCustomFallback(): void {
      this.fingerprintAuth.verifyFingerprintWithCustomFallback({
        message: 'Scan yer finger', // optional
        fallbackMessage: 'Enter PIN' // optional
      }).then(
        () => {
          alert({
            title: "Biometric ID OK",
            okButtonText: "Sweet"
          });
        },
        (error) => {
          alert({
            title: "Biometric ID NOT OK",
            message: (error.code === -3 ? "Show custom fallback" : error.message),
            okButtonText: "Mmkay"
          });
        }
        );
    }`

EddyVerbruggen, 2017-12-13:
Ah, Android. Could be specific to the Android version or the manufacturer's overlay (if any). I have no idea. Anyone?

EddyVerbruggen, 2017-12-14:
I think this is normal behavior actually - please reopen with evidence if it's not.

closed 2017-12-14 by EddyVerbruggen