EddyVerbruggen/nativescript-fingerprint-auth #23: Cannot set property 'useCustomAndroidUI' of undefined

domzinhuu, 2018-02-08:
Im create a app with angular template and when i try use 

verifyFingerprint(options)

i always get the cannot set property useCustoAndroidUI.

my Options:

```
this.verifyOptions.useCustomAndroidUI = false;
this.verifyOptions.authenticationValidityDuration = 50;
this.verifyOptions.message = "coloca ai";
this.verifyOptions.title = "É Tu??";
```

I need set any other setting when using  nativescript angular template?

EddyVerbruggen, 2018-02-08:
Can you share a repo that I can help you with?

domzinhuu, 2018-02-11:
Hi @EddyVerbruggen 

Git repo: https://github.com/domzinhuu/finger

so... this is the error and happen only when i click on de button "On/off" and then i finish the app to restart again. But if i click on the button and only refresh the code on vscode the app restart and work fine.

```
JS: Error in fingerprint-auth.verifyFingerprint: TypeError: Cannot read property 'getSupportFragmentManager' of undefined
JS: ANGULAR BOOTSTRAP DONE. 3182
JS: ERROR Error: Uncaught (in promise): Error: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.Resources$Theme android.content.Context.getTheme()' on a null object reference
JS:     android.app.AlertDialog.resolveDialogTheme(AlertDialog.java:222)
JS:     android.app.AlertDialog$Builder.<init>(AlertDialog.java:452)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1088)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:970)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:957)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:941)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:933)
JS:     com.tns.NativeScriptActivity.onStart(NativeScriptActivity.java:38)
JS:     android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1270)
JS:     android.app.Activity.performStart(Activity.java:6704)
JS:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2641)
JS:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2743)
JS:     android.app.ActivityThread.-wrap12(ActivityThread.java)
JS:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1490)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:154)
JS:     android.app.ActivityThread.main(ActivityThread.java:6165)
JS:     java.lang.reflect.Method.invoke(Native Method)
JS:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
JS:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)
JS: Error: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.Resources$Theme android.content.Context.getTheme()' on a null object reference
JS:     android.app.AlertDialog.resolveDialogTheme(AlertDialog.java:222)
JS:     android.app.AlertDialog$Builder.<init>(AlertDialog.java:452)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1088)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:970)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:957)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:941)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:933)
JS:     com.tns.NativeScriptActivity.onStart(NativeScriptActivity.java:38)
JS:     android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1270)
JS:     android.app.Activity.performStart(Activity.java:6704)
JS:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2641)
JS:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2743)
JS:     android.app.ActivityThread.-wrap12(ActivityThread.java)
JS:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1490)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:154)
JS:     android.app.ActivityThread.main(ActivityThread.java:6165)
JS:     java.lang.reflect.Method.invoke(Native Method)
JS:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
JS:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)
JS:     at createAlertDialog (file:///data/data/org.nativescript.finger/files/app/tns_modules/tns-core-modules/ui/dialogs/dialogs.js:12:17)
JS:     at file:///data/data/org.nativescript.finger/files/app/tns_modules/tns-core-modules/ui/dialogs/dialogs.js:97:27
JS:     at new ZoneAwarePromise (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:777:29)
JS:     at Object.alert (file:///data/data/org.nativescript.finger/files/app/tns_modules/tns-core-modules/ui/dialogs/dialogs.js:94:12)
JS:     at file:///data/data/org.nativescript.finger/files/app/app.component.js:22:29
JS:     at ZoneDelegate.invoke (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:365:26)
JS:     at Object.onInvoke (file:///data/data/org.nativescript.finger/files/app/tns_modules/@angular/core/bundles/core.umd.js:4783:33)
JS:     at ZoneDelegate.invoke (file:///data/data/org.nativescript.fing
JS: er/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:364:32)
JS:     at Zone.run (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:125:43)
JS:     at file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:761:57
JS:     at ZoneDelegate.invokeTask (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:398:31)
JS:     at Object.onInvokeTask (file:///data/data/org.nativescript.finger/files/app/tns_modules/@angular/core/bundles/core.umd.js:4774:33)
JS:     at ZoneDelegate.invokeTask (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:397:36)
JS:     at Zone.runTask (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:165:47)
JS:     at Function.drainMicroTaskQueue (file:///data/data/org.nativescript.finger/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:593:35)
ActivityManager: Displayed org.nativescript.finger/com.tns.NativeScriptActivity: +3s519ms

``

EddyVerbruggen, 2018-02-12:
Hi, thanks for that demo app, but unfortunately I can't reproduce the crash with it. Looking at the code I'd advise to not trigger the fingerprint UI before the page has loaded, so `ngOnInit` is probably not the best place. To check if that's the problem, please test triggering the scanner from a button.

Also, I can't find `useCustomAndroidUI` in your code, so I'm not sure we're looking at the same thing.

domzinhuu, 2018-02-12:
Thanks for the answer, i'll try your tips and i'm back later with result.