EddyVerbruggen/nativescript-fingerprint-auth #24: Verify finger print error

Nurgunkalol, 2018-02-16:
Hello!

 "tns-ios": {
      "version": "3.4.1"
    },
    "tns-android": {
      "version": "3.4.1"
    }
"nativescript-fingerprint-auth": "^4.0.1",

Error in fingerprint-auth.verifyFingerprint: TypeError: Cannot set property 'onActivityResult' of undefined

mariuspruski, 2018-02-21:
Hey.. since nativescript 3.4 they don't define foregroundActivity anymore until navigatedTo event is called, so in ngAfterViewInit it isn't available. See https://github.com/NativeScript/NativeScript/issues/5307

Our workaround is to call the plugin (version 4.0.1) like this:
`this.page.on(Page.navigatedToEvent, () => {
  const fAvailable = await this.fingerprintAuth.available(); 
  ...
});`

EddyVerbruggen, 2018-02-21:
You're right - I've fixed that in 6.0.1.

closed 2018-02-21 by EddyVerbruggen