EddyVerbruggen/nativescript-fingerprint-auth #25: Android build crashes with 6.0.0

mariuspruski, 2018-02-21:
Hi,
Since there are issues with the foreground activity being undefined in NS 3.4.1, I want to update to nativescript-fingerprint-auth. While deploying the app to my Android phone works with version 5.1.0, I'm getting the following error with 6.0.0 and 6.0.1:

`Running full build

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformClassesWithDexForF0F1F2F3F4F5F6F7Debug'.
> com.android.build.api.transform.TransformException: com.android.ide.common.process.ProcessException: java.util.concurrent.ExecutionException: com.android.dex.DexException: Multiple dex files define Lorg/intellij/lang/annotations/Identifier;

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 10s
Unable to apply changes on device: LGH84099771c61. Error is: Command gradlew.bat failed with exit code 1.
Executing after-watch hook from D:\dbk\mobile-app\hooks\after-watch\nativescript-dev-sass.js
Executing after-watch hook from D:\dbk\mobile-app\hooks\after-watch\nativescript-dev-typescript.js`

I have tried cleaning all (platforms remove, app uninstalling,..) but I can't get rid of this error. Any idea?

EddyVerbruggen, 2018-02-21:
I don't see anything related to this plugin in that stacktrace. Feel free to clone this repo and run the demo to see if that works, and reopen with more info:

```bash
git clone https://github.com/EddyVerbruggen/nativescript-fingerprint-auth
cd nativescript-fingerprint-auth/src
npm run demo.android
```

closed 2018-02-21 by EddyVerbruggen

mariuspruski, 2018-03-05:
For anyone looking for a solution for this - I had to excldue the intellij annotations in the app.gradle file:
```
configurations {
  compile.exclude group: 'com.intellij', module: 'annotations'
}
```