EddyVerbruggen/nativescript-fingerprint-auth #27: Plugin not working when deployed via TestFlight/iTunesConnect

mikaelkalt, 2018-03-07:
When using the plugin in a production build deployed via `tns publish ios` the availability check seems to fail.

The code we are running and which works in a normal development build fine, are the following lines of code: 

```
 public ngOnInit() {
       this.authenticate();
    }

    private async authenticate() {
        let fSuccessful: boolean = false;
        const fAvailable: BiometricIDAvailableResult = await this.fingerprintAuth.available();

        if (this.secureStorageService.isFingerprintSettingEnabled() && fAvailable.any) {
            try {
                await this.doVerifyFingerprint();
                fSuccessful = true;
            } catch (e) {
                trace.write("Error during fingerprint authentication: " + e, trace.categories.Debug);
            }
        }

        if (fSuccessful) {
            this.routerExtensions.navigate(["/home"], {clearHistory: true, animated: false});
        } else {
            this.routerExtensions.navigate(["/authenticate"], {clearHistory: true, animated: false});
        }
    }

    private doVerifyFingerprint(): Promise<void> {
        return this.fingerprintAuth.verifyFingerprintWithCustomFallback({
            fallbackMessage: localize("authenticate.scanFallback"),
            message: localize("authenticate.scanFinger"),
        });
    }
```
Since it is somehow very hard to get some information from a TestFlight build about what's happening, I could figure out the following error in the console.app on my mac: 

```
-[LAClient evaluatePolicy:options:uiDelegate:reply:]_block_invoke -> (null), Error Domain=com.apple.LocalAuthentication Code=-1004 "User interaction is required." UserInfo={FingerDatabaseHash=<886d059e ec073f18 ec996a29 27374575 4d1b9c2b 42ecef54 f7b134db 335584f9>, BiometryType=1, AvailableMechanisms=(
    1
), NSLocalizedDescription=User interaction is required.} on <private>
```
The app is not crashing but the availability functionality is not working, it seems to be false although my phone has TouchID enabled and it works fine in development builds locally. 

When downgrading the plugin to version 4.0.1 it works fine for iOS, but there we have the problem of the compatibility on Android with the new nativescript version.

We would like to find a way fixing the problem but currently we're pretty lost. We don't really understand what's the difference between the local build and a TestFlight build, what happens there that could influence the plugin's functionality. 
 
Any suggestions for the analytics or solutions of the problem?

mikaelkalt, 2018-03-09:
I was able to kind of localise the problematic lines of code, so in the `fingerprint-auth.ios.ts` in the availability check you've lately added the additional properties to distinguish between touch and face authentification. I had to remove these lines again to get it to work: 

```
touch: hasBio && laContext.biometryType === LABiometryTypeTouchID,
face: hasBio && laContext.biometryType === LABiometryTypeFaceID, 
```
As we're close to release we needed to quickly fix it, therefore we've forked your plugin. You can see the fix here: 
https://github.com/canoo/nativescript-fingerprint-auth/tree/fix/testFlightAvailabilityIssue

I know this is not a proper solution, just a workaround, as we don't need to know which one is used. Nevertheless, something must be wrong with these two lines, but I don't know what? 

I've even quickly created an XCode project and verified that these lines of code work natively on iOS and they do. It even works in Nativescript locally, it actually only appears when publishing to TestFlight (via  `tns publish ios` command)

What's your opinion on this? I'd be happy to provide a final solution to this, but I'm out of ideas. Since I don't understand the difference between a local production build and a TestFlight release.

EddyVerbruggen, 2018-03-10:
Thanks for those details! I've replaced those properties by their constant values. Should work. Now available in 6.0.2.

closed 2018-03-10 by EddyVerbruggen

mikaelkalt, 2018-03-19:
Thank you very much for fixing this bug, it works fine now! 

Could you maybe explain to me why this only appeared on a TestFlight build? I'd like to understand the problem bit better, as we also had different issue which only appeared on a TestFlight build.

EddyVerbruggen, 2018-03-19:
Looks like a production build strips certain iOS SDK symbols it shouldn't. Like those constants that I've replaced with their runtime values.