EddyVerbruggen/nativescript-fingerprint-auth #30: iOS "verifyFingerprintWithCustomFallback" resolves fail when attempting enter passcode

rickywong-arbaholdings, 2018-03-23:
iOS "verifyFingerprintWithCustomFallback" resolves fail when attempting enter passcode (after a fail attempt of Fingerprint authentication).

Steps to reproduce:
1. Have the Fingerprint authentication UI on screen
2. Authenticate with an incorrect fingerprint, then "fallbackMessage" (i.e. Enter Passcode) shows up
3. Click on the "Enter Passcode" (depends on the string of "fallbackMessage")
4. Resolve as fail right the way (actual passcode screen never shown)

P.S. No issue when using "verifyFingerprint"

FYR
iOS 11.2.1 on 6 Plus
nativescript-fingerprint-auth 6.0.3
tns-ios 3.3.0

CAJazzer, 2018-12-21:
+1 same issue

EddyVerbruggen, 2019-01-05:
That's by design; as the name suggests you should provide your own (custom) fallback when using `verifyFingerprintWithCustomFallback`.

closed 2019-01-05 by EddyVerbruggen