EddyVerbruggen/nativescript-fingerprint-auth #31: Android: minSdkVersion 17 cannot be smaller than version 23

sebawita, 2018-03-23:
hey,

I am trying to add nativescript-fingerprint-auth.

When I build for Android, I am getting this error:
```
* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : uses-sdk:minSdkVersion 17 cannot be smaller than version 23 declared in library [com.github.JesusM:FingerprintManager:v2.0.2] /Users/witalec/.gradle/caches/transforms-1/files-1.1/FingerprintManager-v2.0.2.aar/cbc3c418a7ed90935aaea75c6cf2ad9a/AndroidManifest.xml as the library might be using APIs not available in 17
        Suggestion: use a compatible library with a minSdk of at most 17,
                or increase this project's minSdk version to at least 23,
                or use tools:overrideLibrary="com.jesusm.kfingerprintmanager" to force usage (maylead to runtime failures)
```

my `AndroidManifest.xml` has the following setting:
```
<uses-sdk
android:minSdkVersion="23"
android:targetSdkVersion="__APILEVEL__"/>
```

My runtime is:
```
"tns-android": {
    "version": "4.0.0-rc-2018.3.22.1"
}
```
but I also tried with: 
```
"tns-android": {
"version": "3.4.0"
}
```

I am using `NativeScript@rc`

Any idea what I am doing wrong?

EddyVerbruggen, 2018-03-23:
Can you please try with tns-android 3.4? This may be a bu in 4.0 because the plugin actually does exactly what that error you posted recommends (because I ran into it as well):

https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/b074307a84352eb66e88618aa46c987cd5da80e4/src/platforms/android/AndroidManifest.xml#L7

sebawita, 2018-03-23:
actually... it does work with 3.4.
So this is just a NativeScript issue

EddyVerbruggen, 2018-03-23:
Yeah, I guess we'll need to report this breaking change in the tns-android project.

closed 2018-03-23 by EddyVerbruggen

wanpeng2008, 2018-04-20:
The same issue under nativescript 4.0.0 tns-android 4.0.1

RoyiNamir, 2018-05-15:
@sebawita  Same here . Did you solved / open an issue about it ?

glen-stephens, 2018-05-24:
Same issue here under nativescript 4.0.0 tns-android 4.0.1.  Verified it works under nativescript 3.4.  What is the best workaround until this get fixed? Really don't want to change my minSdkVersion to 23.

mikaelkalt, 2018-05-25:
Another workaround I've just figured out, is to manipulate your `App_Resources/Android/AndroidManifest.xml` manually and adding what @EddyVerbruggen mentioned before: 
`  <uses-sdk tools:overrideLibrary="com.jesusm.kfingerprintmanager"/>`

Important: It needs to be before the the line that defines your minSdkVersion.

glen-stephens, 2018-05-25:
@mikaelkalt Thanks for the help!  I had the tools:overrideLibrary and it wasn't working but I didn't know it needed to be before the minSdkVersion line.  Appreciate it!

kleingtm, 2018-08-17:
Worth noting -- That solution finally worked for me, but I had to blow away the gradle cache:
`rm -rf ~/.gradle/caches`

M-Ali-B, 2018-10-24:
Well my problem was solved when i changed the _minSdkVersion_ in both (module:app) and (module:base) to 23 
My error was because the merger failure of manifest file of (module:app) and (module:base) , the reason was that _minSdkVersion_ was different in both of the files .

ryanammons, 2019-02-20:
Did this actually get reported? It's still an issue, and I don't like having to target minimum 23.