EddyVerbruggen/nativescript-fingerprint-auth #33: Secure lock screen hasn't been set up

r2fer-bit, 2018-05-21:
Hi,
When I set useCustomAndroidUI to true and change the AndroidManifiest.xml, y get the error:
{"code":30,"message":"Secure lock screen hasn't been set up.\n Go to \"Settings -> Security -> Screenlock\" to set up a lock screen."}

But, I test the app with a phone with screen lock setted by password!

If I change useCustomAndroidUI to false, It runs OK!

Please can you help me?

tns --versión: 4.0.1
Android: 6.0.1 (marshmallow)

kandangit, 2019-10-29:
Have  exactly the same issue. With useCustomAndroidUI  set to false, it will work fine (asks for pattern or fingerprint, as setup in Settings->Security->Screenlock), but need to be able to customize the screen. 

Please, how can you change the fingerprint screen?

BlueHunter99, 2019-10-29:
@kandangit what line did you add? I cannot see it.

kandangit, 2019-10-30:
Updated the comment, the line was scrapped as it needed code formatting. Just did that, and a new compile/deploy.

mcalc001, 2020-01-16:
Hey, We are having similar issues. On android, when the user changes (adds/removes) a fingerprint in their phone settings, they can no longer use touch id within the app. This message shows within the console.

```JS: Secure lock screen hasn't been set up.
JS:  Go to "Settings -> Security -> Screenlock" to set up a lock screen. error on biomentricLogin```

It is picking that there is results Bio available.

 {
  "any": true,
  "touch": true
 }

Any Ideas around this? Is this something that can be changed within the app of will the user have to manually update his security settings? 

Thanks!

tns: 6.3

ReeceReynolds, 2020-01-17:
> Hey, We are having similar issues. On android, when the user changes (adds/removes) a fingerprint in their phone settings, they can no longer use touch id within the app. This message shows within the console.
> 
> Secure lock screen hasn't been set up.
> Go to "Settings -> Security -> Screenlock" to set up a lock screen. error on biomentricLogin
> 
> It is picking that there is results Bio available.
> 
> {
> "any": true,
> "touch": true
> }
> 
> Any Ideas around this? Is this something that can be changed within the app of will the user have to manually update his security settings?
> 
> Thanks!
> 
> tns: 6.3

Experiencing the same issue, is this a phone setting or something I can control in app?

paul-castro, 2020-01-24:
@EddyVerbruggen is this a device issue? or the plugin itself?

Any update about this?

gitpf, 2020-03-17:
> Hey, We are having similar issues. On android, when the user changes (adds/removes) a fingerprint in their phone settings, they can no longer use touch id within the app. This message shows within [...]

same issue here. Anyone managed to resolve this somehow? Everything works fine on iOS :/

sandwai-scottjackman, 2020-09-06:
Same problem here, @EddyVerbruggen can we get an update on this please? 
plugin: 7.0.2
tns: 6.8

sandwai-scottjackman, 2020-09-08:
For anyone getting the following message when trying to implement useCustomAndroidUI and getting the following message:

Secure lock screen hasn't been set up.
Go to "Settings -> Security -> Screenlock" to set up a lock screen. error on biomentricLogin

I was able to fix this by removing all of my registered biometric  data from the phone I was testing on. I then re-added a finger print and the correct custom dialogue showed. Hope this helps someone, as I was ready to give up on this.

3rror404, 2021-08-19:
It seems to work if you set the "use_password" string resource to an empty string in this file: https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/master/demo/app/App_Resources/Android/src/main/res/values/strings.xml

`<string name="use_password">Use password</string>`

rubenhr, 2021-10-08:
Have anyone figure out how to use FaceID in Android? I have setup a project, but for android is only returning biometrics but it doesn't return touch or face.?