EddyVerbruggen/nativescript-fingerprint-auth #35: CustomAndroidUI will not use device password to authenticate, here's why

shiv19, 2018-08-05:
https://github.com/JesusM/FingerprintManager/blob/333e4ca4e44d5c5fb725e8926690000d926af880/javasample/src/main/java/com/jesusm/fingerprintmanager/javasample/MainActivity.java#L74

`onSuccessWithManualPassword` was intended to be used when the developer will manually maintain
a separate app password. Which he will compare.
So you would want to `resolve(password)` here, https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/7ab2ddef09204320ae9db8bf40312685d6bfe72b/src/fingerprint-auth.android.ts#L117

but by doing so, the functionality of using password won't be same between using and not using custom UI.

shiv19, 2018-08-05:
This was discovered while debugging @RoyiNamir 's app

EddyVerbruggen, 2018-08-09:
Hey @shiv19, thanks for reporting this!

I always assumed that password entry widget of the custom UI was similar to the one on the default UI, but it turns out the custom UI relies on a custom password the user needs to configure for the app (instead of using the system-wide password).

For a number of reasons, I think the best solution is to just `resolve(password)`. I've added this to the demo app and the readme.

closed 2018-08-09 by EddyVerbruggen

shiv19, 2018-08-09:
Perfect 👌 that should do it :)

RoyiNamir, 2018-08-09:
I must say that it will be very hard for a person to see "Enter password" , and to understand that the password is not the system password - but a custom -compare-to-something password....

EddyVerbruggen, 2018-08-09:
That's up to your app to make clear, I guess.