EddyVerbruggen/nativescript-fingerprint-auth #38: Missing class

aminebizid, 2018-09-13:
Hello,
I get this error on app startup

System.err: java.lang.RuntimeException: Unable to instantiate activity ComponentInfo{com.gem.live/org.nativescript.fingerprintplugin.AppCompatActivity}: java.lang.ClassNotFoundException: **Didn't find class "org.nativescript.fingerprintplugin.AppCompatActivity**" on path: DexPathList[[zip file "/data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/base.apk"],nativeLibraryDirectories=[/data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/lib/arm, /data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib, /product/lib]]
System.err: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3064)
System.err: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3278)
System.err: 	at android.app.ActivityThread.-wrap12(Unknown Source:0)
System.err: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1894)
System.err: 	at android.os.Handler.dispatchMessage(Handler.java:109)
System.err: 	at android.os.Looper.loop(Looper.java:166)
System.err: 	at android.app.ActivityThread.main(ActivityThread.java:7377)
System.err: 	at java.lang.reflect.Method.invoke(Native Method)
System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:469)
System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:963)
System.err: Caused by: java.lang.ClassNotFoundException: Didn't find class "org.nativescript.fingerprintplugin.AppCompatActivity" on path: DexPathList[[zip file "/data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/base.apk"],nativeLibraryDirectories=[/data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/lib/arm, /data/app/com.gem.live-sdOgOMnXlHlx3bvBx5fbwQ==/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib, /product/lib]]
System.err: 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:125)
System.err: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
System.err: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
System.err: 	at android.app.Instrumentation.newActivity(Instrumentation.java:1179)
System.err: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3054)
System.err: 	... 9 more

Nurgunkalol, 2018-10-19:
same with webpack, how did u solved?

Nurgunkalol, 2018-11-15:
up

jdnichollsc, 2019-05-03:
Very odd this issue

EddyVerbruggen, 2019-05-04:
Can anyone share how to reproduce this please?