EddyVerbruggen/nativescript-fingerprint-auth #42: Execution failed for task ':app:copyMetadata'.

SergeyMell, 2018-12-05:
I'm getting this crash while trying to run application for android

    Execution failed for task ':app:copyMetadata'.

My minSDK version is 17 and I've added all the necessary changes
    <manifest
        xmlns:android="http://schemas.android.com/apk/res/android"
        xmlns:tools="http://schemas.android.com/tools"
        package="__PACKAGE__" android:versionCode="205" android:versionName="0.2.5">
    <supports-screens android:smallScreens="true" android:normalScreens="true" android:largeScreens="true"
                      android:xlargeScreens="true"/>
    <uses-sdk android:minSdkVersion="17"
              android:targetSdkVersion="__APILEVEL__"
              tools:overrideLibrary="com.jesusm.kfingerprintmanager"/>

My tns core module version is 4.2.0

EddyVerbruggen, 2018-12-05:
Are you able to run the demo app successfully?

SergeyMell, 2018-12-05:
On demo app build goes ok, but I get a crash while running the appication

```
An uncaught Exception occurred on "main" thread.
java.lang.RuntimeException: Unable to instantiate activity ComponentInfo{org.nativescript.fingerprintauth/org.nativescript.fingerprintplugin.AppCompatActivity}: java.lang.ClassNotFoundException: Didn't find class "org.nativescript.fingerprintplugin.AppCompatActivity" on path: DexPathList[[zip file "/data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/base.apk"],nativeLibraryDirectories=[/data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/lib/arm, /data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/base.apk!/lib/armeabi-v7a, /system/lib, /system/vendor/lib]]
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2679)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
	at android.app.ActivityThread.-wrap11(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6501)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.ClassNotFoundException: Didn't find class "org.nativescript.fingerprintplugin.AppCompatActivity" on path: DexPathList[[zip file "/data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/base.apk"],nativeLibraryDirectories=[/data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/lib/arm, /data/app/org.nativescript.fingerprintauth-e0GsmQnrCkbT5RyDyC6GZw==/base.apk!/lib/armeabi-v7a, /system/lib, /system/vendor/lib]]
	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:125)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
	at android.app.Instrumentation.newActivity(Instrumentation.java:1175)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2669)
	... 9 more
```

EddyVerbruggen, 2018-12-05:
1. What steps did you take to run that?
2. And is there a repo I can look at for your app's issue?

SergeyMell, 2018-12-05:
1.
npm i
tns prepare android
tns run android

2.
No, unfortunately...

SergeyMell, 2018-12-05:
Yes, demo application works as expected being run with `npm run demo.android`

EddyVerbruggen, 2018-12-05:
OK, so since you can't share your repo please compare the demo with your project yourself.

closed 2018-12-05 by EddyVerbruggen

SergeyMell, 2018-12-05:
Yeah, seems the issue is somewhere inside my code or configurations. Thank you for your time