EddyVerbruggen/nativescript-fingerprint-auth #43: Android 7.0 not working

robtoll, 2018-12-18:
I'm testing on 3 Android devices, none of which have touchID functionality
the function "fingerprintAuth.available()" doesnt return anything on a Sony Xperia - L1, Android 7.0
it works fine on a Sony Z3 (Android 6.01) and an HTC Sense (Android 4.4.2)
Also works for my iPhone and iPad.
Any ideas why Android 7.0 would have problems?
Thanks

EddyVerbruggen, 2018-12-19:
Perhaps the error is swallowed. Do you have a `catch` clause, similar to this?

```typescript
    this.fingerprintAuth.available()
        .then((result: BiometricIDAvailableResult) => {
          console.log("doCheckAvailable result: " + JSON.stringify(result));
        })
        .catch(err => {
          console.log("doCheckAvailable error: " + err);
        });
```

robtoll, 2018-12-19:
Hmm, yes - this works as a fallback, but I thought it was supposed to return an object that shows the available support - whether it is available or not.
It's OK tho - I can use it in this way, I juyst have to code it a bit differently.
Thanks

EddyVerbruggen, 2018-12-19:
I was actually interested in the error you caught because indeed that’s not what’s supposed to happen.

robtoll, 2018-12-19:
Oh, sorry, I misunderstood!
The error outputs a string "Device doesn't support fingerprint authentication"
Thanks for your support Eddy

closed 2019-06-26 by EddyVerbruggen