EddyVerbruggen/nativescript-fingerprint-auth #46: iOS Passcode fallback never presented after failed TouchID

CAJazzer, 2019-02-02:
Using the Demo app on my iOS device,  the "verify with passcode fallback" call simply reverts to the catch alert() in doVerifyFingerprint() after 3 failed attempts using touch.  No err is displayed (undefined), only the title and okButtonText.
my iOS version: 12.1.3

EddyVerbruggen, 2019-02-12:
I just tested the embedded demo app on my iPad Pro and it behaves as expected: after failing to scan the finger 3 times the promise is rejected. Can you give it a try?

CAJazzer, 2019-02-13:
Thanks,

Same results.  

I downloaded (cloned) the demo, then used Xcode to set a valid certificate and run the project on my iPhone 6s & on my iPad Air. 

The problem is I never get the passcode screen.

Using the “verify with passcode” button, after 3 tries all I get is the alert dialog:

Biometric ID NOT
OK / canceled
------------------
    Mmkay
    
    
"Never get the passcode fallback"

marcocasadio, 2019-03-11:
hi, the same for me

bradmartin, 2019-06-12:
I looked into this. Might need to use the `DeviceOwnerAuthentication` of `LAPolicy` for the passcode fallback. Not 100% certain on this ATM.

EddyVerbruggen, 2019-06-26:
I've just tested with Face ID on my iPhone X and it seems to work as expected. The only thing I can think of is [this check failing](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/e6e737a946b8323ab055dcad6563e616b901a5de/src/fingerprint-auth.ios.ts#L85) so it fell through to the 'verify with custom callback' method.

So I've applied a small change where this fall through will still trigger the passcode fallback by applying the policy @bradmartin suggested.

closed 2019-06-26 by EddyVerbruggen