EddyVerbruggen/nativescript-fingerprint-auth #47: is there a way to return a unique biometric ID associated with the fingerprint?

sputn1k, 2019-02-22:
for example? - 
```
fingerprintAuth.available().then(
        function(avail) {
            fingerprintAuth.verifyFingerprint({
                  message: "Scan with your finger",
                  fallbackTitle: "Enter PIN"
              }).then(function() {
                  console.log("return....unique biometric ID?!"); // how do i return it? 
              }, function(error) {
                  console.log("error.....");
              });
        }
)
```

EddyVerbruggen, 2019-02-22:
No. That's not available in the SDKs. Check fi. Apple's docs on Touch ID.

closed 2019-02-22 by EddyVerbruggen

sputn1k, 2019-02-22:
How do we associate the account with the touch authentication?

EddyVerbruggen, 2019-02-22:
That's up to you(r app logic). But the bottom line is: only 1 user can use touchid to log in. I'm pretty sure you can't find an app where multiple profiles can log in with their own fingerprints.

sputn1k, 2019-02-22:
@EddyVerbruggen thanks eddy I get you. cheers