EddyVerbruggen/nativescript-fingerprint-auth #56: Nativescript 6 Update error-  Unable to instantiate activity ComponentInfo{org.nsf.mobileaudit/org.nativescript.fingerprintplugin.AppCompatActivity}:

acharyaks90, 2019-07-09:
After Updating the app to version 6. Not able to debug or build 
tns debug android --bundle 

System.err: Unable to instantiate activity ComponentInfo{org.nsf.mobileaudit/org.nativescript.fingerprintplugin.AppCompatActivity}: java.lang.ClassNotFoundException: Didn't find class "org.nativescript.fingerprintplugin.AppCompatActivity" on path: DexPathList[[zip file "/data/app/org.nsf.mobileaudit-z67-cJAhPFGRLXmJChBdQA==/base.apk"],nativeLibraryDirectories=[/data/app/org.nsf.mobileaudit-z67-cJAhPFGRLXmJChBdQA==/lib/arm64, /data/app/org.nsf.mobileaudit-z67-cJAhPFGRLXmJChBdQA==/base.apk!/lib/arm64-v8a, /system/lib64, /system/vendor/lib64]]

EddyVerbruggen, 2019-07-09:
Do you have `AppCompatActivity` in your AndroidManifest.xml?

acharyaks90, 2019-07-11:
@EddyVerbruggen  Yes i have

EddyVerbruggen, 2019-07-11:
That's the problem then - you'll need to remove that because I've managed to automate that step with more recent NativeScript versions. Please check out https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/tree/d453e12323c83f941fe236190dc8bb2286056a97#%EF%B8%8F-important-note-when-using-nativescript--540

closed 2019-07-11 by EddyVerbruggen

acharyaks90, 2019-07-11:
@EddyVerbruggen Thanks