EddyVerbruggen/nativescript-fingerprint-auth #57: CustomUI for android not triggered after keyboard enter

bbroereES, 2019-07-09:
Using the plugin (7.0.1) on Nativescript angular (5.4.0) with custom android UI:
When the user chooses to use pin instead of fingerprint, enters the pincode and does *not* press 'OK' in the dialog but the enter key on the keyboard, the dialog gets closed but the callback of verifyFingerprint is never called.

larxenne, 2019-12-12:
I have the exact same issue, did you find any solution ?

bbroereES, 2019-12-13:
Yes, I did :)
See this [feature request](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/issues/58):
> This can be achieved by setting the use_password string in strings.xml to empty. The button to cancel will not be shown. Request to update readme to inform others.

larxenne, 2019-12-13:
I tried it, but this remove the cancel button.
If I chose the password option and then use the enter key  on the keyboard instead of the 'Ok' button, the callback is still not triggered.

bbroereES, 2019-12-13:
Could reproduce, don't see an immediate fix though... For me the solution was to just remove the fallback option, but you are right, the callback is not triggered.

larxenne, 2019-12-13:
Yes, actually with the 'Cancel' button, you fallback in the catch.
The enter key is not triggering anything.

That's sad, i cannot use the customandroidui, I guess i l use the basic one