EddyVerbruggen/nativescript-fingerprint-auth #59: Broken router?

przpl, 2019-07-10:
I have following method:

`public async verify() {
        const available = await this._fingerprintAuth.available();
        if (!available || !available.touch) {
            return; // in my test the method does not return because available.touch is true as expected
        }

        try {
            await this._fingerprintAuth.verifyFingerprint({});
        } catch (error) { // in my test method does not throw exception, authentication is successful
            alert("Fingerprint not recognized");
            return;
        }
       
        const result = await this._routerExtensions.navigateByUrl("some-url"); // this line of code is executed and the result is true

        // workaround, explanation below
        this.showNextButton = true;
    }

public onNextBtnTap() {
        this._routerExtensions.navigateByUrl("some-url"); // this works
}
`

The problem is that navigation does not work, nothing happens. If I remove lines associated with fingerprintAuth the navigation works. What's wrong in this case? How the library impacts the router? I have to workaround this problem by showing "Next" button after successful authentication. User has to tap the button but I would like navigate automatically without user interaction.

bbroereES, 2019-07-10:
Try navigating after a delay of say 2 seconds to see if this is a timing issue.

przpl, 2019-07-10:
It solved the problem. Thanks :) But why it works that way?

bbroereES, 2019-07-10:
Well, the closing of the modal takes time and apparently during that time routing does not seem to work all the time :)

closed 2019-07-10 by EddyVerbruggen