EddyVerbruggen/nativescript-fingerprint-auth #61: Cancel event

patrikengborg, 2019-09-05:
After invoking the `verifyFingerprint` function and the user presses cancel - How do I catch that cancel event / press?

Thanks.

patrikengborg, 2019-09-10:
@EddyVerbruggen any idea? Would really appreciate some help on this one. Thanks.

EddyVerbruggen, 2019-09-10:
Can you start by mentioning which platform(s) and perhaps share a repo I can look at?

EddyVerbruggen, 2019-09-11:
Hi, that's very clear indeed, thanks! I don't think it's possible for me to know whether that function failed or was cancelled, but perhaps if you use `verifyFingerprintWithCustomFallback` you may get the relevant details in the `reject` (or `catch` in your case) because of [this line](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/d453e12323c83f941fe236190dc8bb2286056a97/src/fingerprint-auth.ios.ts#L145).. can you give that a shot?

patrikengborg, 2019-09-11:
Yes!

`{ code: -2, message: 'Canceled by user.' }`

Brilliant, thank you so much!

closed 2019-09-11 by patrikengborg