EddyVerbruggen/nativescript-fingerprint-auth #64: Touch ID strange behavior IOS13

yonkoGH, 2019-09-29:
Hi Eddy 

I'm facing a strange behavior with this awesome plugin of yours after upgrading my 
iPhone7 OS to IOS 13.1.1

The Touch ID dialog doesn't show directly after a Button Click (for example)
Sometimes I need to make another action to my phone screen to make the dialog appears  
or sometimes  the Touch ID dialog shows twice .

debug error

> evaluatePolicy:options:uiDelegate:reply:]_block_invoke -> (null), Error Domain=com.apple.LocalAuthentication Code=-1004 "User interaction is required." UserInfo={BiometryType=1,

gabrielbiga, 2019-09-30:
In my application the fingerprint dialog doesn't show at all in iOS 13+.

EddyVerbruggen, 2019-09-30:
Do you experience this with the demo app in this repo as well? If not, can you share a demo? And wondering if this also happens to you with FaceID?

yonkoGH, 2019-09-30:
Hi Eddy 
----
I tried the demo app with my iPhone 7 (Touch ID) 
The demo is a little bit buggy
the Touch dialog is taking  much longer and sometimes it doesn't show at all 
and i have to press the button twice.

Also i'm getting this message again 
` Error Domain=com.apple.LocalAuthentication Code=-1004 "User interaction is required." UserInfo={BiometryType=1,
`

laschaffer, 2019-09-30:
Hi Eddy
I can confirm that the finger and face-id auth in the same app with iOS13 does not work correctly.
Sometimes the native dialog appears after a click sometime after 2x click or it does not appear but the authentication is active (so I can enter touching the fingerprint without seeing the dialog)

EddyVerbruggen, 2019-09-30:
I just updated the demo. Perhaps the alerts were hampering performance. It's lightning fast for me (FaceID on iPhone 11 Pro). Can you re-clone and try again? And if there's still an issue please reopen and share a video of the problem.

closed 2019-09-30 by EddyVerbruggen

laschaffer, 2019-10-02:
Hi 
I think there is somewhere a problem yet, please see this video (iPad 3, iOS 13.x.x latest )
I have made some changes in my code so you can better seeing what is happening

igorgiovannini, 2020-06-05:
The issue is still reproducible with the latest iOS versions (iOS 13.5, 13.5.1).
Is there a workaround?