EddyVerbruggen/nativescript-fingerprint-auth #66: ios13 problem

laschaffer, 2019-10-02:
Hi 
I think there is somewhere a problem yet, please see this video (iPad 3, iOS 13.x.x latest )

https://drive.google.com/file/d/10_ZjhV7hBPmID5iZUBmEHlLQiu-PsZAS/view?usp=drivesdk

I have made some changes in my code so you can better seeing what is happening
I click at the button -> the image background is being flashed indicating the action ,after 1.5 sec the verifyFingerprint will be called -> now the native popup should be opened but nothing happens till I reach the home button with my finger -- I tried several times and sometimes the popup is visible after the 1.5 sec delay but only very rare
....
  private activateScan() {
    try {
      this.biometricButtonState = true;
      this.biometricScanActive = true;
      this.scanTimerh = setInterval(()=> {
          this.biometricButtonState = !this.biometricButtonState;
      },500);
      setTimeout(() => {
        this.verifyFingerPrint();
      },isIOS ?  1500 : 0);
    }catch(error) {
      console.log(`onBiometricStackLoaded:${error}`);
      this.feedback.error({
        message: error
      })
    }

   this.fingerprintAuth.verifyFingerprint(
      {
        message: 'Scan your finger', // optional (used on both platforms) - for FaceID on iOS see the notes about NSFaceIDUsageDescription
        title: 'Biometric authentication',
        authenticationValidityDuration: 60, // optional (used on Android, default 5)
        // fallbackMessage: 'Enter Domain Password', // optional, the button label when scanning fails (default: 'Enter password')
        useCustomAndroidUI: true
      })
      .then((enteredPassword) => {
....

_Originally posted by @laschaffer in https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/issues/64#issuecomment-537409812_

bbroereES, 2019-10-04:
We have encountered this issue aswel. Fortunately (in a way) this is not an issue of the plugin, but rather a bug in iOS 13, 13.1 and 13.1.1. See the following article for example:
https://www.forbes.com/sites/kateoflahertyuk/2019/09/30/warning-as-apple-ios-13-touch-id-bug-hits-millions-of-iphone-users/#4fb549454fa4

laschaffer, 2019-10-07:
ok thanks

closed 2019-10-07 by laschaffer