EddyVerbruggen/nativescript-fingerprint-auth #67: Build falied with exit code 1

delanick, 2019-10-06:
Hey @EddyVerbruggen 

I'm probably being daft, because I did add the manifest file stuff regarding minimum API and target, but still getting a broken build. Here are my logs, please advise if I've just missed something - happy to share more with you :)

`[19-10-06 14:36:16.514] Build started.
[19-10-06 14:36:16.527] (CLI) Copying template files...
[19-10-06 14:36:18.645] (CLI) Platform android successfully added. v6.0.2
[19-10-06 14:36:18.647] (CLI) Preparing project...
[19-10-06 14:36:21.571] (CLI) File change detected. Starting incremental webpack compilation...
[19-10-06 14:36:21.606] (CLI) webpack is watching the files…
[19-10-06 14:36:25.303] (CLI) Hash: 9f02e4e7b6955099e229
[19-10-06 14:36:25.303] (CLI) Version: webpack 4.27.1
[19-10-06 14:36:25.303] (CLI) Time: 3718ms
[19-10-06 14:36:25.303] (CLI) Built at: 10/06/2019 2:36:25 PM
[19-10-06 14:36:25.303] (CLI)                                      Asset       Size   Chunks             Chunk Names
[19-10-06 14:36:25.303] (CLI)                                  bundle.js   81.4 KiB   bundle  [emitted]  bundle
[19-10-06 14:36:25.303] (CLI) fonts/Font Awesome 5 Duotone-Solid-900.otf   1.59 MiB           [emitted]
[19-10-06 14:36:25.303] (CLI)     fonts/Font Awesome 5 Pro-Light-300.otf   1.09 MiB           [emitted]
[19-10-06 14:36:25.303] (CLI)                      fonts/FontAwesome.ttf    162 KiB           [emitted]
[19-10-06 14:36:25.303] (CLI)            fonts/MaterialIcons-Regular.ttf    125 KiB           [emitted]
[19-10-06 14:36:25.303] (CLI)                               package.json  112 bytes           [emitted]
[19-10-06 14:36:25.303] (CLI)                                 runtime.js   71.4 KiB  runtime  [emitted]  runtime
[19-10-06 14:36:25.303] (CLI)                        tns-java-classes.js    0 bytes           [emitted]
[19-10-06 14:36:25.303] (CLI)                                  vendor.js   4.54 MiB   vendor  [emitted]  vendor
[19-10-06 14:36:25.303] (CLI) Entrypoint bundle = runtime.js vendor.js bundle.js
[19-10-06 14:36:25.303] (CLI) [./ sync ^\.\/app\.(css|scss|less|sass)$] . sync nonrecursive ^\.\/app\.(css|scss|less|sass)$ 174 bytes {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./ sync recursive (?<!\bApp_Resources\b.*)\.(xml|css|js|(?<!\.d\.)ts|(?<!\b_[\w-]*\.)scss)$] . sync (?<!\bApp_Resources\b.*)\.(xml|css|js|(?<!\.d\.)ts|(?<!\b_[\w-]*\.)scss)$ 613 bytes {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./app-root.xml] 312 bytes {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./app.css] 856 bytes {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./app.js] 2.22 KiB {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./camera/camera-page.js] 4.27 KiB {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./camera/camera-page.xml] 1.26 KiB {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./camera/camera-view-model.js] 544 bytes {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./home/home-page.js] 2.18 KiB {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./home/home-page.xml] 782 bytes {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./home/home-view-model.js] 532 bytes {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./shared/dialogObjects.js] 532 bytes {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./shared/endpoint_management.js] 631 bytes {bundle} [built]
[19-10-06 14:36:25.303] (CLI) [./shared/global_variables.js] 393 bytes {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI) [./shared/pushMessaging.js] 4.81 KiB {bundle} [optional] [built]
[19-10-06 14:36:25.303] (CLI)     + 285 hidden modules
[19-10-06 14:36:25.519] (CLI) Webpack compilation complete. Watching for file changes.
[19-10-06 14:36:25.520] (CLI) Webpack build done!
[19-10-06 14:36:25.533] (CLI) Project successfully prepared (android)
[19-10-06 14:36:26.629] (CLI) Starting Cloud build. Cloud operation id: e13b32d9-3a67-4890-9303-8d5d53e79db0
[19-10-06 14:36:26.629] (CLI) Getting accounts information...
[19-10-06 14:36:28.102] (CLI) Using account 05fbadec-4fd6-4eae-b628-5adf1ae57352.
[19-10-06 14:36:28.102] (CLI) Cloud build of '/Users/nickdelahunt/Documents/OneDrive/wigital/Mobile apps/KitchenSink', platform: 'Android', configuration: 'Debug'.
[19-10-06 14:36:29.385] Upload step started.
[19-10-06 14:36:29.689] (CLI) Unable to use git, reason is:
[19-10-06 14:36:29.689] (CLI) Command /usr/bin/git failed with exit code 69 Error output:
[19-10-06 14:36:29.689] (CLI) Agreeing to the Xcode/iOS license requires admin privileges, please run “sudo xcodebuild -license” and then retry this command.
[19-10-06 14:36:50.893] Upload step completed.
[19-10-06 14:36:52.096] Build step started.
[19-10-06 14:36:59.287] (CLI) GRADLE_USER_HOME is /opt/docker_shared/ybGlXb11AZ0b7i5zim.gradle
[19-10-06 14:36:59.287] (CLI) android Build Tooling revision 2019.08.21.1r
[19-10-06 14:36:59.287] (CLI) Build Tooling Hostname 2300a3e5b044
[19-10-06 14:36:59.287] (CLI) Unzip Project Files
[19-10-06 14:36:59.287] (CLI) NativeScript CLI version 6.0.3
[19-10-06 14:36:59.287] (CLI) Add platform android with runtime version 6.0.2
[19-10-06 14:37:01.121] (CLI) Copying template files...
[19-10-06 14:37:03.670] (CLI) Platform android successfully added. v6.0.2
[19-10-06 14:37:14.752] (CLI) Preparing project...
[19-10-06 14:37:18.992] (CLI) Hash: 180d61208976f8bd6c64
[19-10-06 14:37:18.992] (CLI) Version: webpack 4.27.1
[19-10-06 14:37:18.992] (CLI) Time: 2379ms
[19-10-06 14:37:18.992] (CLI) Built at: 10/06/2019 12:43:16 PM
[19-10-06 14:37:18.992] (CLI)                                      Asset       Size   Chunks             Chunk Names
[19-10-06 14:37:18.992] (CLI)                                  bundle.js   81.4 KiB   bundle  [emitted]  bundle
[19-10-06 14:37:18.992] (CLI) fonts/Font Awesome 5 Duotone-Solid-900.otf   1.59 MiB           [emitted]
[... 103 lines cut ...]
* What went wrong:
A problem occurred evaluating project ':app'.
> The prefix "tools" for attribute "tools:overrideLibrary" associated with an element type "uses-sdk" is not bound.
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
* Get more help at https://help.gradle.org
BUILD FAILED in 17s
Command ./gradlew failed with exit code 1
'Build failed with error code 2'.
[19-10-06 14:38:16.191] Stopped LiveSync on 52108bf447e0c3ad for /Users/nickdelahunt/Documents/OneDrive/wigital/Mobile apps/KitchenSink.
[19-10-06 14:38:16.193] Couldn't enable debugging for 52108bf447e0c3ad, Error: Couldn't enable debugging for 52108bf447e0c3ad
at Errors.fail (/usr/local/lib/node_modules/nativescript/lib/common/errors.js:126:28)
at Errors.failWithoutHelp (/usr/local/lib/node_modules/nativescript/lib/common/errors.js:136:21)
at DebugController.<anonymous> (/usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:116:30)
at Generator.next (<anonymous>)
at /usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:13:71
at new Promise (<anonymous>)
at __awaiter (/usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:9:12)
at DebugController.enableDebuggingCoreWithoutWaitingCurrentAction (/usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:113:16)
at DebugController.enableDebuggingCoreWithoutWaitingCurrentAction (/usr/local/lib/node_modules/nativescript/lib/common/decorators.js:67:47)
at DebugController.<anonymous> (/usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:209:25)
at Generator.next (<anonymous>)
at fulfilled (/usr/local/lib/node_modules/nativescript/lib/controllers/debug-controller.js:10:58)
at process._tickCallback (internal/process/next_tick.js:68:7)
`

EddyVerbruggen, 2019-10-09:
Hi!

That error indicates this line is missing:

https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/38c09eb976782a5c01d8b4edad885e408a1305b7/demo/app/App_Resources/Android/src/main/AndroidManifest.xml#L3

closed 2019-10-09 by EddyVerbruggen

delanick, 2019-10-12:
@EddyVerbruggen it worked, thanks bud! :)