EddyVerbruggen/nativescript-fingerprint-auth #71: On setting useCustomAndroidUI in Android, popup button color seems to be broken

ThilinaTCH, 2019-12-03:
I'm using v7.0.2 with Nativescript 6
Please check image 
[image]

ThilinaTCH, 2019-12-06:
@EddyVerbruggen Any idea on this issue, please

EddyVerbruggen, 2019-12-06:
Perhaps there's a generic CSS class in your app that applies that color to the button? Or maybe it's one of [these](https://github.com/EddyVerbruggen/nativescript-fingerprint-auth/blob/master/demo/app/App_Resources/Android/src/main/res/values/colors.xml).

ThilinaTCH, 2019-12-06:
@EddyVerbruggen Yeah we checked them and same colors there in our app, can't find any style overriding them

EddyVerbruggen, 2019-12-06:
Did you check the `com.github.JesusM:FingerprintManager` package (Google it)? We're using v2.0.2 in this plugin. Maybe their project repo has some docs/related issues.

ThilinaTCH, 2019-12-09:
@EddyVerbruggen OK sure will check there, Thanks!

ThilinaTCH, 2019-12-09:
@EddyVerbruggen I checked with that package and seems no issue there as they haven't changed or upgraded it recently. Don't know whether it causes with Nativscript 6 as there it upgrades to Androidx, anyone experienced the same?

EddyVerbruggen, 2019-12-09:
I thought maybe in that package there's a way to influence the UI by (perhaps) some XML config files.