EddyVerbruggen/nativescript-fingerprint-auth #73: fingerprint-auth.available: TypeError: Cannot read property 'isHardwareDetected' of null

cloudhx, 2020-01-15:
When running my app on a new Android 9 device (previously running on Android 7 without this problem), it reports the following error:

`fingerprint-auth.available: TypeError: Cannot read property 'isHardwareDetected' of null`

Method `isHardwareDetected` was added in `API level 23` and deprecated in `API level 28` which is Android 9 Pie, I was thinking this might be an issue for API level 28 and later versions.

TPham92, 2020-05-06:
Any updates on this issue?

Pandishpan, 2020-05-11:
I'm experiencing the same issue on Android 9,  Phone Model SM-A105FN (PPR1.180610.011) (this model doesn't have a fingerprint sensor?)

Uncaught (in promise): TypeError: Cannot read property 'isHardwareDetected' of null
TypeError: Cannot read property 'isHardwareDetected' of null

However, I can't replicate the same issue on a phone with the same Android version but **with** a fingerprint sensor. I think this is happening if the phone doesn't have a fingerprint sensor and the API level is 28 or higher.

cloudhx, 2020-05-24:
Thanks @Pandishpan I think it makes sense, I will check my devices and also see if there is a way to catch the exception in this case.

erjdriver, 2020-05-24:
+1.

Phone is without fingerprint sensor but with faceid.