EddyVerbruggen/nativescript-fingerprint-auth #74: fingerprint duplicate dialog appears

nikitakulish, 2020-01-30:
Samsung s10, Android 9
"nativescript-fingerprint-auth": "^7.0.2",
useCustomAndroidUI: true

Please check it.
[image]

jamesst20, 2020-04-28:
My guess here is that your phone is having an in-display fingerprint scanner. 

The whitebox behind is the "useCustomAndroidUI" however your phone has its own fingerprint dialog. The right fix would be to have useCustomAndroidUI to false on devices having an in-display fingerprint scanner

khurram-gp, 2020-11-05:
@jamesst20 Is there any way to know that a device have `in-display fingerprint scanner`?