EddyVerbruggen/nativescript-fingerprint-auth #75: Face unlock

hariaravind10, 2020-01-31:
Hi @EddyVerbruggen 

Is this plug supports to unlock the app using face id In Android Devices? 

In the native script-fingerprint-auth  Documentation face id supports in only iPhone 11.

EddyVerbruggen, 2020-01-31:
Correct, this plugin only supports iOS. I don't currently have an Android device that supports face unlock, so I can't really help there.

antonio92cop, 2020-02-28:
Hi @hariaravind10 
In this plug not work face unlock in android because use deprecated library FingerprintManager.

timoschlueter, 2020-05-19:
@EddyVerbruggen do you have any plans on migrating this plugin to the BiometricPrompt API which Google introduced as a replacement for FingerprintManager?

EddyVerbruggen, 2020-05-21:
Not personally, but a PR will be merged of course.

gwojnar, 2020-11-22:
Hello, 

@EddyVerbruggen I will also ask for migration plans.
FingerprintManager is deprecated as of API 28 and replaced by BiometricPrompt. I see some problems on my phone with Android 10 - fingerprint dialog not working.
I tried to build a native application using BiometricPrompt and it works fine.